m5stack/TimerCam-arduino #6: Build Failure on Arduino

Strvm, 2021-08-12:
Hello, I'm trying to test out the TimerCam X Camera using Arduino on a mac, but when building I get this compile error:

`/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c: In function 'bat_init':
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c:10:25: error: 'ADC1_GPIO38_CHANNEL' undeclared (first use in this function); did you mean 'ADC1_CHANNEL_7'?
 #define BAT_ADC_CHANNEL ADC1_GPIO38_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c:26:31: note: in expansion of macro 'BAT_ADC_CHANNEL'
     adc1_config_channel_atten(BAT_ADC_CHANNEL, ADC_ATTEN_DB_11);
                               ^~~~~~~~~~~~~~~
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c:10:25: note: each undeclared identifier is reported only once for each function it appears in
 #define BAT_ADC_CHANNEL ADC1_GPIO38_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c:26:31: note: in expansion of macro 'BAT_ADC_CHANNEL'
     adc1_config_channel_atten(BAT_ADC_CHANNEL, ADC_ATTEN_DB_11);
                               ^~~~~~~~~~~~~~~
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c: In function 'bat_get_adc_raw':
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c:10:25: error: 'ADC1_GPIO38_CHANNEL' undeclared (first use in this function); did you mean 'ADC1_CHANNEL_7'?
 #define BAT_ADC_CHANNEL ADC1_GPIO38_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c:40:25: note: in expansion of macro 'BAT_ADC_CHANNEL'
     return adc1_get_raw(BAT_ADC_CHANNEL);
                         ^~~~~~~~~~~~~~~
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c: In function 'bat_get_voltage':
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c:10:25: error: 'ADC1_GPIO38_CHANNEL' undeclared (first use in this function); did you mean 'ADC1_CHANNEL_7'?
 #define BAT_ADC_CHANNEL ADC1_GPIO38_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
/Users/strum/Documents/Arduino/libraries/Timer-CAM/src/battery.c:46:39: note: in expansion of macro 'BAT_ADC_CHANNEL'
         adc_raw_value += adc1_get_raw(BAT_ADC_CHANNEL);
                                       ^~~~~~~~~~~~~~~
exit status 1`

When looking up on the internet I found this issue on another repo that had a similar issue: https://github.com/m5stack/M5EPD/issues/10 . Any ideas?

elParaguayo, 2021-08-15:
I just bought one of these and ran into the same issue.

I added `#include "soc/adc_channel.h"` to `battery.c` and it compiles. Haven't had a chance to upload yet.

Strvm, 2021-08-16:
> I just bought one of these and ran into the same issue.
> 
> I added `#include "soc/adc_channel.h"` to `battery.c` and it compiles. Haven't had a chance to upload yet.
> 
> EDIT: confirmed - this works.

Hey, alright cool to know. For me I got it working by modifying **battery.c**:

from this:
`#define BAT_ADC_CHANNEL ADC1_GPIO38_CHANNEL`

to this:
`#define BAT_ADC_CHANNEL ADC1_CHANNEL_2`

Tinyu-Zhao, 2022-08-10:
Thank you for your feedback, the problem has now been fixed.

closed 2022-08-10 by Tinyu-Zhao