m5stack/TimerCam-arduino #10: Problem with a compilance

clf75, 2022-01-04:
Hi Guys

Sorry for my english ;-)

We have a new M5tstack-timer and I will save my wifi for this camera can be connect by my wifi but, when I will do compilance with arduino software , we have a issues

> Arduino:1.8.19 (Windows Store 1.8.57.0) (Windows 10), Tarjeta:"M5Stack-Timer-CAM, Enabled, Default(3MB No OTA/1MB SPIFFS), 240MHz (WiFi/BT), 1500000, None"
> 
> C:\Users\**********\Documents\Arduino\libraries\TimerCam-arduino-master\src\app_httpd.cpp:22:10: fatal error: fd_forward.h: No such file or directory
> 
>  #include "fd_forward.h"
> 
>           ^~~~~~~~~~~~~~
> 
> compilation terminated.
> 
> Se encontraron varias bibliotecas para "WiFi.h"
> 
> Usado: C:\Users\********\Documents\ArduinoData\packages\m5stack\hardware\esp32\2.0.1\libraries\WiFi
> 
>  No usado: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi
> 
> exit status 1
> 
> Error compilando para la tarjeta M5Stack-Timer-CAM.
> 

Thanks!!

danielgeiser, 2022-01-20:
Hi, I have the same issue.
Please help. 
Thx

vnikonov63, 2022-02-01:
/Users/elbrus/Documents/Arduino/libraries/Timer-CAM/src/app_httpd.cpp:22:10: fatal error: fd_forward.h: No such file or directory
Multiple libraries were found for "WiFi.h"
 #include "fd_forward.h"
 Used: /Users/elbrus/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.2/libraries/WiFi
          ^~~~~~~~~~~~~~
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
compilation terminated.
exit status 1
Error compiling for board M5Stack-Timer-CAM.

Please help have the same issue.

clf75, 2022-02-01:
I returned the product to the seller

konacurrents, 2022-08-09:
Ok, I have the same issue. It looks like the fd_forward and fr_forward are for FACE recognition (which I'm not as interested in at this moment). QR code decoding would be more valuable.

Any ideas on where these are? I'll have to edit out them in the source to see if I can get past that error. 

I just received 2 of the units and am excited to get them working with my same ESP32 and M5 code base, configured appropriately.

thanks.

Tinyu-Zhao, 2022-08-10:
I'm sorry I'm only replying now, which program are you compiling to get this error, I need more information to reproduce the problem, please provide all the information you need to compile if you can.

konacurrents, 2022-08-10:
Hi. I'm trying the web_cam from the examples. I grabbed all the code and commented out the fd_forward.h ref and it runs nicely, WiFi address viewing the camera. 

I'll invite you to a little GitHub with your code. 

Do you have code that reads and decides QR codes? I could use that. 

-scott in Seattle.

Tinyu-Zhao, 2022-08-10:
> Hi. I'm trying the web_cam from the examples. I grabbed all the code and commented out the fd_forward.h ref and it runs nicely, WiFi address viewing the camera.
>
> I'll invite you to a little GitHub [...]
You can try adding this library.
https://github.com/alvarowolfx/ESP32QRCodeReader

closed 2022-08-23 by Tinyu-Zhao