m5stack/TimerCam-arduino #12: Release request for version 0.0.3

708yamaguchi, 2022-08-23:
Thank you very much for maintaining the highly useful TimerCam firmware!
I too have been able to successfully compile `web_cam` firmware with the latest commit.
https://github.com/m5stack/TimerCam-arduino/commit/eabb74fb56a87bc1373b1c25bc924560fa6d1e04

When you have time, could you release version 0.0.3 for the arduino library?
I was having a hard time getting it to compile with the current latest version 0.0.2.
Thank you.

Tinyu-Zhao, 2022-08-23:
OK, thanks for the tip, I've posted it and will probably be able to update it in a few hours

closed 2022-08-23 by Tinyu-Zhao

slim-bean, 2022-09-03:
The arduino libs haven't picked up 0.0.3 yet, I think maybe it needs also to be updated in library.properties and library.json?

Tinyu-Zhao, 2022-09-08:
> The arduino libs haven't picked up 0.0.3 yet, I think maybe it needs also to be updated in library.properties and library.json?

Sorry, my negligence has been re-released.