m5stack/TimerCam-arduino #15: How to use the "ON"-Button

v8-rt, 2023-05-24:
Hi guys,
Sorry if the question sounds stupid... but i don't find any hint, to which GPIO, the "ON"-Button on the Side of the TimerCam, is connected to. On other M5-Modules the PWR-Button isn't even connected to the esp32, but to another chip that also controls the RTC.
Do you know, how i can use it either as trigger to take photos, or turn on/off the camera?
Thank you in advance!

felmue, 2023-06-03:
Hello @v8-rt 

it's connected to GPIO37. See Pinmap/Button [here](https://docs.m5stack.com/en/unit/timercam).

It is my understanding that when the camera is powered off it will power it on. And when the camera is already powered on it depends on what you program your application to do when GPIO37 is pulled down to GND.

Thanks
Felix

vollukas, 2024-11-21:
Thanks @FedericoBusero for help! I would also love to learn more about how TimerCAMs power work and more working examples would be nice. I want to make pictures more often and I'd like to use powerbank with my unit, but timerSleep() doesn't wake up when there is power through USB port..