m5stack/TimerCam-arduino #19: The device doesn't wake up from sleeping

raduprv, 2024-04-01:
### Describe the bug

I am using the following code, from the example:
    TimerCAM.begin(true);
    TimerCAM.Power.timerSleep(5);

I placed it at the end of my program, the device goes to sleep but never wakes up. I placed it at the beginning, before doing anything else (except for printing a serial essage), and the same thing happens.
The device is powered by USB from the computer, no battery attached.

Am I doing something wrong?

### To reproduce

Using Arduino

### Expected behavior

I would expect the device to wake up after 5 seconds.

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

imliubo, 2024-04-02:
Hi,the hardware design need unplug the USB cable and connect the battery,if you connect the USB cable it won't go sleep,because the power not cut off.

And why need unplug the USB cable,i think you need check the [schematic](https://docs.m5stack.com/en/unit/timercam_x).

raduprv, 2024-04-02:
Thank you for replying. Perhaps you should state that somewhere on the main page of the camera, or at least make a comment in source code sleep example, because this is not very intuitive :)

imliubo, 2024-04-02:
Yes,you are right, we will add notice to example and the docs.
Thanks for your advice.

raduprv, 2024-04-02:
Great, thank you!
And to not open a new issue, is there any way I could add a SD card to the Timer Camera? There are two pins on the connector, and there are some testing pads on the left and right side of the board. The left side has tx/rx, but on the right side I couldn't understand what they are for. Are they connected to any GPIO? Can any of them be used for this purpose? That would be very useful, as right now I can only upload files to my computer by ftp, one by one.