m5stack/TimerCam-arduino #20: Using Camera as I2C Slave

pacmac, 2024-04-02:
### Describe the bug

I am using the camera as a I2C slave, all of the communication is working.

As the master is battery powered, it also needs to power the cam, I tried doing that using the 5V supply on the connector, but when I put it to deep sleep, the power consumption is very high.

As I understand that the ultra low power consumption is only available when on battery power, I am now trying to power using the battery connector, but the problem is the camera does not even power on when supplied by battery.

So how can I supply power via the battery connector and have the device switch on without needing to press the physical switch.

I have tried setting GPIO33 high and low but that makes no difference, the cam wont power up when supplied by battery ?

Thanks

### To reproduce

See above

### Expected behavior

See above

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

pacmac, 2024-04-02:
Hi, thanks for helping I did see your post in the forum before I posted the issue here.

Unfortunately as far as I can see, that wont work for me, as the camera is a slave to another ESP32 Master which is doing a lot of other things, and one of them is to take a photo.

The RTC would only work if the TimerCam controls itself and decides when to wake up, but I need to have it take a photo on demand not when it wants to.

pacmac, 2024-04-03:
Hi @imliubo - can it be triggered with a RTC Wakeup pin ?

Gitshaoxiang, 2024-04-08:
the  ultra low power mode actually mens turn off all power supply.  in this situation only two way to wakeup the device(RTC wakeup pin signal, or Wakeup button.),  but if you connected the 5V supply to the ext port. the TimerCAM will nerver shutdown. even you set the device to deepsleep mode(it maybe still have 20ma consumption),  I think you could try to add a mos switch to control the Camera 5V power supply.

pacmac, 2024-04-08:
Thanks, yes that's what I did.

closed 2024-04-08 by pacmac