m5stack/TimerCam-arduino #21: TimerCAM needs an end/stop method

raduprv, 2024-04-02:
### Describe the bug

So my usecase is simple:
I need to check on the battery voltage. If it's ok, I do camera stuff, then I put it to sleep.
However, the camera doesn't seem to work after using TimerCAM.begin(). I am getting i2c errors, and others have gotten them too.

So there should be an end method, so that people can do stuff like read the voltage, end the TimerCAM stuff, use the camera, end the camera, use TimerCAM.begin again, and so on.

Right now, I can't check the voltage at the beginning if I want to sue the camera after that.

### To reproduce

Tested on Arduino, but it should be the same all over.

### Expected behavior

Be able to terminate TimerCAM

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Gitshaoxiang, 2024-04-07:
thanks for you feedback , now it updated. could you pull the lib update, then test again?

```cpp
#include "M5TimerCAM.h"
#include <WiFi.h>

void led_breathe(int ms) {
    for (int16_t i = 0; i < 255; i++) {
        TimerCAM.Power.setLed(i);
        vTaskDelay(pdMS_TO_TICKS(ms));
    }

    for (int16_t i = 255; i >= 0; i--) {
        TimerCAM.Power.setLed(i);
        vTaskDelay(pdMS_TO_TICKS(ms));
    }
}

void setup() {
    TimerCAM.begin(true);

    if (!TimerCAM.Camera.begin()) {
        Serial.println("Camera Init Fail");
        return;
    }
    Serial.println("Camera Init Success");

    TimerCAM.Camera.sensor->set_pixformat(TimerCAM.Camera.sensor,
                                          PIXFORMAT_JPEG);
    TimerCAM.Camera.sensor->set_framesize(TimerCAM.Camera.sensor,
                                          FRAMESIZE_SVGA);
    TimerCAM.Camera.sensor->set_vflip(TimerCAM.Camera.sensor, 1);
    TimerCAM.Camera.sensor->set_hmirror(TimerCAM.Camera.sensor, 0);

    Serial.printf("Bat Voltage: %dmv\r\n", TimerCAM.Power.getBatteryVoltage());
    Serial.printf("Bat Level: %d%%\r\n", TimerCAM.Power.getBatteryLevel());

    if (TimerCAM.Camera.deinit()) {
        Serial.println("Camera Deinit.");
    }

    if (!TimerCAM.Camera.begin()) {
        Serial.println("Camera Init Fail");
        return;
    }
    Serial.println("Camera Init Success");

    Serial.printf("Bat Voltage: %dmv\r\n", TimerCAM.Power.getBatteryVoltage());
    Serial.printf("Bat Level: %d%%\r\n", TimerCAM.Power.getBatteryLevel());
    led_breathe(10);
    TimerCAM.Power.timerSleep(5);
}

void loop() {
}

```

raduprv, 2024-04-07:
Thank you for replying. I am not using the camera class provided by TimerCAM, I am using the 'standard' way, like the ESP32-Cam camera code, so that it can be compatible with all the boards, and keep the board specific stuff (like sleeping) to a minimum.

Gitshaoxiang, 2024-04-08:
yes,  actually the TimerCAM class is using the standard API.  also in this [commit](https://github.com/m5stack/TimerCam-arduino/commit/03f811d661314af82c5d42b4a71e6060aaf19cd1#diff-4e5d1406649edd23aa5d39d7b9afd2128d47bddc7a761255c2b98b71781c4451), we update the I2C port.  I think you could modify the same thing then try again.

raduprv, 2024-04-09:
Can you make a new release too, please?